sinara-hw/EEM_FMC_Carrier #23: Servmod voltage levels

filipswit, 2022-01-24:
Servmod signal can have either 3.3v logic or 2.5v logic.
The problem is when it is driven from 3.3v and it's enable signal for IC21 which is powered from 2.5v.

marmeladapk, 2022-02-01:
Is servmod from backplane always a P3V3 signal?

Additionally servmod generated from IC24 at P2V5 drives IC16, IC27-30 which operate at P3V3.

So, servmod should be split to servmod_2V5 and servmod_3V3 and voltage translation should be 2V5->3V3 in EEM variant and 3V3->2V5 in DIOT variant.

marmeladapk, 2022-02-01:
@filipswit Could you check if these changes are correct?

filipswit, 2022-02-05:
> Is servmod from backplane always a P3V3 signal?

While it's driven from DIOT SB it's P3V3

> So, servmod should be split to servmod_2V5 and servmod_3V3 and voltage translation should be 2V5->3V3 in EEM variant and 3V3->2V5 in DIOT variant.

Now it's better.

closed 2022-02-05 by filipswit