sinara-hw/EEM_FMC_Carrier #50: FMC Clock Configuration for Shuttler DACs Analog Clock Source

linuswck, 2023-09-06:
For the shuttler DACs, the AD9117 analog clock(CLKIN pin) should get a clean clock source, preferably not from FPGA pins. 

However, (at least on the production version of my board) both FMC clock is connected to FPGA pins.
FMC_CLK_BDIR_P2/N2: C210, C211 are placed
FMC_CLK_BDIR_P3/N3: C215, C216 are placed

In the production version, either FMC_CLK_BDIR_P2/N2 or FMC_CLK_BDIR_P3/N3 should be configured to connect IC1.

gkasprow, 2023-09-06:
It was done on purpose to support carriers that produce clean clocks, for example MTCA AFCZ/AFC/ACK/AFCKU

gkasprow, 2023-09-06:
FMC carrier  supporta distribution of clean clock and that option should be enabled by default

adbr8389, 2023-09-15:
@gkasprow I am probably missing something obvious; what is the benefit of distributing the clock signal indirectly through the FPGA to the shuttler DAC's instead of directly from the clock buffer? The buffer looks to have several more channels that are unused in its current configuration.

marmeladapk, 2023-09-15:
In FMC standard these clocks are bidirectional, meaning that other FMC cards may generate clock signal on them to the FPGA (e.g. clock alongside data in ADCs).

closed 2024-01-08 by maciejprzybysz