sinara-hw/EEM_FMC_Carrier #53: New revision of EEM FMC Carrier

maciejprzybysz, 2023-09-14:
We are going to make new revision of EFC. Issues to fix:

- [ ] #38
- [ ] #39
- [ ] #40
- [ ] #43
- [ ] #44
- [ ] #45
- [ ] #46
- [ ] #47
- [ ] #48
- [ ] #49
- [ ] #51
- [ ] #52

TBC: 

- [ ] #50

Estimated completion: 10/2023

Feel free to add any suggestions/issues/ideas for new revision.

pvantrap, 2023-11-15:
Hello, would it be possible to add P4 (and maybe P3 for v2, low priority) to keep it compatible with the [DI/OT crate](https://ohwr.org/project/diot-crate-hw/wikis/home) RTM connections? We're interested in using this board at CERN SY-ABT as a cheaper alternative for https://ohwr.org/project/diot-pfc-ku/wikis/home

marmeladapk, 2023-11-15:
@pvantrap This would require changing the FPGA. There's not enough pins on the current one.

gkasprow, 2023-11-15:
We use it currently with [Shuttler](https://github.com/sinara-hw/Shuttler/wiki) and [Waver](https://github.com/sinara-hw/Waver_FMC/wiki) FMCs.
We have plans to use it with [FMC TDC ADC](https://github.com/elhep/FMC_ADC100M_10B_TDC_16cha/wiki) for CERN AEGIS project

gkasprow, 2023-11-15:
AFAIR guys from Oxford use it with FMC CoaxPress cards to capture images from sCMOS cameras.

gkasprow, 2023-11-16:
We may also expose I2C or pins some from I2C extender...

M1ch4U, 2023-11-16:
The new revision v1.1 (rc1) of EEM_FMC_Carrier has been published. The project is ready for community review - if you are interested in this project feel free for posting any remarks/comments.

ChangeLOG
- Fixed parts availability ISSUE #38
- In FPGA_SDRAM.SchDoc NCP51145PDR2G changed to NCP51200MNTXG	ISSUE #39
- In EEM_Connectors_JTAG.SchDoc I2C enabling IC added 	ISSUE #40
- In Power_Supplies.SchDoc IC14 enabled in EEMvariant, R278 changed to Schottky diode.  ISSUE #43
- R80 and R79 changed to 1k, added DIP switch ISSUE #44
- In IC2 pins 6 and 7 disconected ISSUE #45
- In IC10 GPB7 and GPB0 swapped ISSUE #46
- Board shape change, left LEDs SMA mounting holes, right connectors, J10, J11 moved, SMA changed to delock  #47
- R3 changed to 1k ISSUE #48
- Board extended from panel side, LD15 changed ISSUE #49
- Migration from 100T to 200T ISSUE #51
- C103 changed to 680u
- 4.7uF caps in VCCINT increased amount from 6 to 12
- 470nF caps in VCCINT increased amount from 8 to 14
- 470nF caps in VCCBRAM increased amount to from 2 to 3
- Pin delays in 200T added
- FMC position fixed ISSUE #52
- LVDS diffpairs adjusted
- GND vias added near LVDS that changes reference
- vias moved under L26
- On L2 polygon under diff pairs changed (12V, 3V3)
- Impedance control added
- DDR trace width unified to meet impedance control requirements
- Front panel 3D model updated
- Variants updated
- PDN simulation added

maciejprzybysz, 2024-01-22:
EFC v1.1 was released, so I close issue.

closed 2024-01-22 by maciejprzybysz