sinara-hw/EEM_FMC_Carrier #60: add high speed SFP link

gkasprow, 2024-10-04:
several users asked us for high speed SFP link
we can do it as a mezzanine and SATA connector
we can easily fit SFP and route clk in as a pigtail
[image]

marmeladapk, 2025-02-04:
Should I remove the front panel clock signal completely since we have internal mmcx on the back or leave this signal as mmcx near the front?

marmeladapk, 2025-02-04:
Also, currently we have 4 GTP lanes available on this FPGA. In EEM variant all are routed to FMC. In DIOT variant 1 of those can be rerouted to the backplane. Should I add another mux for this signal?

gkasprow, 2025-02-05:
> Should I remove the front panel clock signal completely since we have internal mmcx on the back or leave this signal as mmcx near the front?

yes, one mmcx is enough; we can always use SMA pigtail to route it to the panel

marmeladapk, 2025-02-13:
I have mixed feelings about using assembly option here. It would either mean keeping two stocks of PCBs or sometimes expecting users to solder 0201 capacitors. Both are unrealistic IMO.

I feel like the following connections are a better choice, since these muxes are quite cheap.

```
FPGA MGT2 --- MUX --- SFP
               |----- FMC MGT 2

FPGA MGT3 --- MUX --- BP MGT
               |----- FMC MGT 3
```

For comparison, this is one possible diagram with a capacitor assembly option.
```
FPGA MGT3 --- MUX --- BP MGT
               |------------- Capacitors --- SFP
                                   |-------- FMC MGT 3
```

closed 2025-05-21 by marmeladapk