sinara-hw/EEM_FMC_Carrier #67: V1.2: The AC coupling caps before FMC may not be necessary

MorganTL, 2025-06-17:
I can only found ANSI/VITA 57.1-2008 online which mentioned the AC coupling caps should NOT be on the carrier. However, I cannot access the latest FMC ANSI/VITA 57.1-2019 to doublecheck, as it's behind a paywall.

kaolpr, 2025-07-18:
Let's say I believe it is clearly stated that no components should be placed between FPGA and the connector.

gkasprow, 2025-07-19:
But it also says: 

[image]

[image]

jordens, 2025-07-19:
I.e. AC coupling always and exclusively at and near the RX end.

gkasprow, 2025-07-19:
Not always, the SFP requires that all AC coupling be inside the SFP module, but that makes sense. There are DC-coupled GPON plugins...